OpenDroneMap/ODM #1686: Handle Altum-PT Thermal band

gnosys-scrawford, 2023-08-17:
### How did you install ODM? (Docker, installer, natively, ...)?

Latest Docker image pulled from dockerhub.

### What is the problem?

Processing Altum-PT data with thermal band, when using ```--radiometric-calibration camera+sun```, software crashes.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Expect it to either leave this channel alone, or, convert the values from milliKelvin to Celsius.

In opendm/thermal.py in the dn_to_temperature function, change an "if" statement (line 38) so that it will handle both Altum models correctly:

```if photo.camera_make == “MicaSense” and photo.camera_model == “Altum”:```
to
```if photo.camera_make == “MicaSense” and “Altum” in photo.camera_model:```

This will solve this specific problem, adding support for Altum-PT.
To make the code more robust, the "else" case (line 57) for all unrecognized camera makes and models still is not functioning well and may need attention. (see attached log).

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Attached one Altum-PT thermal band image if you are able to test this function directly.
Otherwise full dataset available at https://shepard-altumpt-test.s3.us-west-2.amazonaws.com/  Ensure that channel 6 files have been removed before running, as they cause ODM to fail in a different manner.  Run equivalent of ```docker run -ti --rm -v C:/Users/scrawford/Documents/data:/datasets my_odm --project-path /datasets subset --dsm --dtm --pc-las --pc-quality high --radiometric-calibration camera+sun```

[IMG_0010_7.zip](https://github.com/OpenDroneMap/ODM/files/12371019/IMG_0010_7.zip)
[odm_crash_log.txt](https://github.com/OpenDroneMap/ODM/files/12371048/odm_crash_log.txt)

gnosys-scrawford, 2023-09-29:
I just deleted this two days ago.  I'll see if I can bring it back.

gnosys-scrawford, 2023-10-02:
I put the data back in the same place for now.

pierotofy, 2023-10-03:
I can't seem to be able to access the files:

```
aws s3 cp s3://shepard-altumpt-test/ altumpt --recursive --region us-west-2
fatal error: An error occurred (AccessDenied) when calling the ListObjectsV2 operation: Access Denied
```

pierotofy, 2023-10-03:
Meh, silly AWS. Here's a complete image list. 

[img_list.txt](https://github.com/OpenDroneMap/ODM/files/12794328/img_list.txt)

PR #1702 (2023-10-04): Altum-PT support
Should close #1686
closed 2023-10-04 by pierotofy