OpenDroneMap/ODM #1687: Possible Bug: pc-geometric breaking reconstruction on non-georeferenced data

Saijin-Naib, 2023-08-19:
### How did you install ODM? (Docker, installer, natively, ...)?
Any
### What is the problem?
Users are reporting that reconstruction can fail with Type 1 / Strange Values when ```--pc-geometric``` (default on, now) is used against datasets without georeference information.

https://community.opendronemap.org/t/better-results-on-surfaces-with-pre-odm-3-0-0/15547/12?u=saijin_naib

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
If ```--pc-geometric``` can be used without georeference, it needs to be fixed. If it can not, it needs to be disabled during the appropriate stage with a log warning.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).
Pending

OTT-Drones, 2023-08-19:
Console log
[iron1console.txt](https://github.com/OpenDroneMap/ODM/files/12386822/iron1console.txt)

Files here: https://drive.google.com/file/d/1vj_gu3Q95bWM3Diw3LSX8NbW-tLUA8-8/view?usp=drive_link

pierotofy, 2023-08-21:
The issue with `pc-geometric` causing problems is probably unrelated to the non-georeferenced nature of datasets, but more likely to the shape of the subjects in the scene.