OpenDroneMap/ODM #1688: Particular image causes WebODM to fail

GaryStebbins, 2023-08-20:
### How did you install ODM? (Docker, installer, natively, ...)?

Native WebODM on Windows 10

### What is the problem?

In the attached ZIP file, image DJI_0767.JPG causes WebODM to fail. Log file shows details. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

As near as I can tell, there is nothing wrong with file DJI_0767.JPG. WebODM should not choke on it. 

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Default processing, including DJI_0767 in file set. 

[WebODM Image Failure.zip](https://github.com/OpenDroneMap/ODM/files/12387251/WebODM.Image.Failure.zip)

pierotofy, 2023-08-22:
Thanks @GaryStebbins , can you post a link to the entire dataset? (25 images) :pray:

pierotofy, 2023-09-29:
Closing as we cannot reproduce this.

closed 2023-09-29 by pierotofy