OpenDroneMap/ODM #1691: Pdal is broken on big VLRs

bugracoskun, 2023-08-25:
Hello everyone,
ODM is broken on large GCP files. I used 17 gcp point and then error. But wehn i tried the without gcp it is working. 
[image]

I searched the error of the pdal. And i found it, it just supports for 65535 bytes for version between 1 and 1.3. It must be used v1.4 for large gcp files.
[image]

Pdal use version 1.2 as a default.
[image]

It can be changed in ODM to version 1.4 after some tests.

bugracoskun, 2023-08-25:
It makes sense. Will it change the accuracy of the data if we don’t use VLRs?

pierotofy, 2023-08-25:
Nop.

bugracoskun, 2023-08-29:
I think this is not size of the our file. PDAL creates VLRs values with our file. As I stated 24KB gcp_list.txt file returns with error. As far as I think it is necessary to determine the limit size for our file.

DiegoContelli, 2023-08-30:
Hello, I'm having the same problem, when I run the process without GCP, with the GCP file the same message appears, there are 21 GPS points, the drone used is a Phantom 3 Pro, approximately 2500 images.
Could you help me to solve this, I'm using WebODM Natively, how do I skip the step? 
Excuse my English, I'm from Brazil. thanks in advance for the help!

DiegoContelli, 2023-09-01:
Hello again, I'm still having this unresolved problem, I tried to remove the --writers.las.vlrs parameter, and another different failure appeared, in another test with only 5 points in the GCP file I was successful in the processing, I created another file with other points and I was successful again now with 6 points, when I create a larger file the error remains.

[INFO] running pdal translate -i "E:\WebODM\resources\app\apps\NodeODM\data\33f214b3-9265-4206-9048-3d7c4f3ee797\odm_filterpoints\point_cloud.ply" -o "E:\WebODM\resources\ app\apps\NodeODM\data\33f214b3-9265-4206-9048-3d7c4f3ee797\odm_georeferencing\odm_georeferenced_model.laz" ferry transformation --filters.ferry.dimensions="views => UserData" --filters.transformation.matrix="1 0 0 569132.0 0 1 0 7469006.0 0 0 1 0 0 0 0 1" --writers.las.offset_x=569132.0 --writers.las.offset_y=7469006.0 --writers.las.scale_x=0.001 --writers.las.scale_y =0.001 --writers.las.scale_z=0.001 --writers.las.offset_z=0 --writers.las.a_srs="+proj=utm +zone=22 +south +datum=WGS84 +units=m +no_defs + type=crs" --writers.las.vlrs="{\"filename\": \"E:/WebODM/resources/app/apps/NodeODM/data/33f214b3-9265-4206-9048-3d7c4f3ee797/odm_georeferencing/ground_control_points .geojson\", \"user_id\": \"ODM\", \"record_id\": 1, \"description\": \"Ground Control Points (GeoJSON)\"}"
PDAL: writers.las: Can't write VLR with user ID/record ID = ODM/1. The data size exceeds the maximum supported

[image]
[image]
[image]

bugracoskun, 2023-09-02:
Hello, what is the size of your gcp_list.txt file? Can you check it?

DiegoContelli, 2023-09-02:
Hi my friend, the size of my gcp_list.txt file is 32KB, I'm in another attempt to process without the --writers.las.vlrs parameter, I performed a new test without the parameter, it worked, but I tested this one with less pictures , I left the total running now with all the photos, return after finishing.

bugracoskun, 2023-09-02:
32KB is too large. Can u test it with 10-15 kb around. You need to delete half of them.

PR #1697 (2023-09-08): Compress GCP data before VLR inclusion
Should fix #1691 by first compressing the GCP data, and then checking that the size is within margins for VLR inclusion.
closed 2023-09-08 by pierotofy