OpenDroneMap/ODM #1707: texrecon link error: undefined reference to 'tbb::

wdstar1, 2023-10-10:
When I was doing the "Linking CXX executable texrecon" step, the following error occurred:

CMakeFiles/texrecon.dir/texrecon.cpp.o: In function `main.cold.462':
/usr/include/tbb/task_scheduler_init.h:132: undefined reference to `tbb::task_scheduler_init::terminate()'
CMakeFiles/texrecon.dir/texrecon.cpp.o: In function `main':
/usr/include/tbb/task_scheduler_init.h:126: undefined reference to `tbb::task_scheduler_init::initialize(int, unsigned long)'
/usr/include/tbb/task_scheduler_init.h:132: undefined reference to `tbb::task_scheduler_init::terminate()'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tbb::detail::d1::concurrent_vector<unsigned long, tbb::detail::d1::cache_aligned_allocator<unsigned long> >::reserve(unsigned long) [clone .part.1100]':
/usr/local/include/oneapi/tbb/cache_aligned_allocator.h:64: undefined reference to `tbb::detail::r1::cache_line_size()'
/usr/local/include/oneapi/tbb/cache_aligned_allocator.h:64: undefined reference to `tbb::detail::r1::cache_line_size()'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tbb::detail::d1::concurrent_vector<unsigned long, tbb::detail::d1::cache_aligned_allocator<unsigned long> >::reserve(unsigned long) [clone .part.1100]':
/usr/local/include/oneapi/tbb/detail/_segment_table.h:520: undefined reference to `tbb::detail::r1::throw_exception(tbb::detail::d0::exception_id)'
/usr/local/include/oneapi/tbb/detail/_segment_table.h:235: undefined reference to `tbb::detail::r1::throw_exception(tbb::detail::d0::exception_id)'
/usr/local/include/oneapi/tbb/detail/_segment_table.h:299: undefined reference to `tbb::detail::r1::throw_exception(tbb::detail::d0::exception_id)'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tbb::detail::d1::concurrent_vector<unsigned long, tbb::detail::d1::cache_aligned_allocator<unsigned long> >::reserve(unsigned long) [clone .part.1100]':
/usr/local/include/oneapi/tbb/cache_aligned_allocator.h:54: undefined reference to `tbb::detail::r1::cache_aligned_allocate(unsigned long)'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tbb::detail::d1::concurrent_vector<unsigned long, tbb::detail::d1::cache_aligned_allocator<unsigned long> >::reserve(unsigned long) [clone .part.1100]':
/usr/local/include/oneapi/tbb/concurrent_vector.h:503: undefined reference to `tbb::detail::r1::throw_exception(tbb::detail::d0::exception_id)'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tex::view_selection(SparseTable<unsigned int, unsigned short, float> const&, UniGraph*, tex::Settings const&)':
/usr/local/include/oneapi/tbb/task_group.h:268: undefined reference to `tbb::detail::r1::initialize(tbb::detail::d1::task_group_context&)'
../../libs/tex/libtex.a(view_selection.cpp.o): In function `tex::view_selection(SparseTable<unsigned int, unsigned short, float> const&, UniGraph*, tex::Settings const&
........................
......................
apps/texrecon/CMakeFiles/texrecon.dir/build.make:125: recipe for target 'apps/texrecon/texrecon' failed
make[5]: *** [apps/texrecon/texrecon] Error 1
CMakeFiles/Makefile2:194: recipe for target 'apps/texrecon/CMakeFiles/texrecon.dir/all' failed
make[4]: *** [apps/texrecon/CMakeFiles/texrecon.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/mvstexturing.dir/build.make:110: recipe for target 'mvstexturing/stamp/mvstexturing-build' failed
make[2]: *** [mvstexturing/stamp/mvstexturing-build] Error 2
CMakeFiles/Makefile2:516: recipe for target 'CMakeFiles/mvstexturing.dir/all' failed
make[1]: *** [CMakeFiles/mvstexturing.dir/all] Error 2
Makefile:83: recipe for target 'all' failed

How can I solve this problem

pierotofy, 2023-10-10:
Are you building with docker? If not, try using docker.

closed 2023-10-13 by pierotofy

wdstar1, 2023-10-16:
> Are you building with docker? If not, try using docker.
No, I tried to deploy it directly on ubantu18.04, but the above problem occurred