OpenDroneMap/ODM #1715: Can't sign up for the community forum

careck, 2023-11-01:
### How did you install ODM? (Docker, installer, natively, ...)?

irrelevant

### What is the problem?

I'm trying to sign up to the ODM community forum at https://community.opendronemap.org/c/webodm/6

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

After clicking the Sign-Up link I get a form where I can fill in my details, email, username and password. After all checks out (green check) I click the sign-up link. 

At this point I expected a success message. However I got this message:

[image]

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

see above description of how I tried to sign up.

careck, 2023-11-01:
I can assure you that I've never registered with this forum before and that the email should not be already registered. 

I also tried to sign up using the Google method, this also did not work.

pierotofy, 2023-11-01:
Ping @smathermather

Saijin-Naib, 2023-11-01:
You can contact hello@uav4geo.com so I can assist you in private as I'll need to verify your email and account name.

careck, 2023-11-01:
> You can contact [hello@uav4geo.com](mailto:hello@uav4geo.com) so I can assist you in private as I'll need to verify your email and account name.

Thanks, I've sent you an email.

closed 2023-11-02 by pierotofy