OpenDroneMap/ODM #1718: Document appropriate rerun-from places when changing settings

smathermather, 2023-11-07:
Per: https://community.opendronemap.org/t/changed-flag-now-where-do-i-restart-from/18126, it's difficult for even some advanced users to know what point to re-run from when trying out different settings. Per João's request, we could establish this as a table, but I lean toward instead/also documenting this in the command documentation themselves so it's always visible.

smathermather, 2023-11-08:
https://github.com/OpenDroneMap/ODM/pull/1719

pierotofy, 2023-11-08:
| **Option**                              | **Restart Stage**     |
| --------------------------------------- | --------------------- |
| **3d-tiles**                            | PostProcess           |
| **align**                               | Georeferencing        |
| **auto-boundary**                       | Point Filtering       |
| **auto-boundary-distance**              | Point Filtering       |
| **bg-removal**                          | Load Dataset          |
| **boundary**                            | Point Filtering       |
| **build-overviews**                     | Orthophoto            |
| **camera-lens**                         | Load Dataset          |
| **cameras**                             | Load Dataset          |
| **cog**                                 | DEM                   |
| **copy-to**                             | PostProcess           |
| **crop**                                | Georeferencing        |
| **dem-decimation**                      | DEM                   |
| **dem-euclidean-map**                   | DEM                   |
| **dem-gapfill-steps**                   | DEM                   |
| **dem-resolution**                      | DEM                   |
| **dsm**                                 | DEM                   |
| **dtm**                                 | DEM                   |
| **end-with**                            | N/A                   |
| **fast-orthophoto**                     | Point Filtering       |
| **feature-quality**                     | Structure From Motion |
| **feature-type**                        | Structure From Motion |
| **force-gps**                           | Structure From Motion |
| **gcp**                                 | Load Dataset          |
| **geo**                                 | Load Dataset          |
| **gltf**                                | Texturing             |
| **gps-accuracy**                        | Load Dataset          |
| **help**                                | N/A                   |
| **ignore-gsd**                          | Structure From Motion |
| **matcher-neighbors**                   | Structure From Motion |
| **matcher-order**                       | Structure From Motion |
| **matcher-type**                        | Structure From Motion |
| **max-concurrency**                     | N/A                   |
| **merge**                               | Merge                 |
| **mesh-octree-depth**                   | Meshing               |
| **mesh-size**                           | Meshing               |
| **min-num-features**                    | Structure From Motion |
| **no-gpu**                              | N/A                   |
| **optimize-disk-space**                 | N/A                   |
| **orthophoto-compression**              | Orthophoto            |
| **orthophoto-cutline**                  | Orthophoto            |
| **orthophoto-kmz**                      | Orthophoto            |
| **orthophoto-no-tiled**                 | Orthophoto            |
| **orthophoto-png**                      | Orthophoto            |
| **orthophoto-resolution**               | Orthophoto            |
| **pc-classify**                         | DEM                   |
| **pc-copc**                             | Georeferencing        |
| **pc-csv**                              | Georeferencing        |
| **pc-ept**                              | Georeferencing        |
| **pc-filter**                           | Multi View Stereo     |
| **pc-las**                              | Georeferencing        |
| **pc-quality**                          | Structure From Motion |
| **pc-rectify**                          | DEM                   |
| **pc-sample**                           | Point Filtering       |
| **pc-skip-geometric**                   | Multi View Stereo     |
| **primary-band**                        | Load Dataset          |
| **project-path**                        | N/A                   |
| **radiometric-calibration**             | Structure From Motion |
| **rerun**                               | N/A                   |
| **rerun-all**                           | N/A                   |
| **rerun-from**                          | N/A                   |
| **rolling-shutter**                     | Structure From Motion |
| **rolling-shutter-readout**             | Structure From Motion |
| **sfm-algorithm**                       | Structure From Motion |
| **sfm-no-partial**                      | Structure From Motion |
| **skip-3dmodel**                        | Meshing               |
| **skip-band-alignment**                 | Structure From Motion |
| **skip-orthophoto**                     | Orthophoto            |
| **skip-report**                         | Report                |
| **sky-removal**                         | Load Dataset          |
| **sm-cluster**                          | Split                 |
| **sm-no-align**                         | Split                 |
| **smrf-scalar**                         | DEM                   |
| **smrf-slope**                          | DEM                   |
| **smrf-threshold**                      | DEM                   |
| **smrf-window**                         | DEM                   |
| **split**                               | Split                 |
| **split-image-groups**                  | Split                 |
[... cut ...]
| **tiles**                               | DEM                   |
| **use-3dmesh**                          | Texturing             |
| **use-exif**                            | Load Dataset          |
| **use-fixed-camera-params**             | Structure From Motion |
| **use-hybrid-bundle-adjustment**        | Structure From Motion |
| **version**                             | N/A                   |
| **video-limit**                         | Load Dataset          |
| **video-resolution**                    | Load Dataset          |

jprates, 2023-11-09:
"Merge" and "Split" are not currently available options in WebODM's "Restart" button.
Assuming we must start from the previous available option, which ones would them be for each case?

Unless of course there is a plan to add them to the available options as well...

pierotofy, 2023-11-09:
@jprates please let's move all questions over to the forum at https://community.opendronemap.org Thank you! :+1:

pierotofy, 2023-11-09:
You seem to make a lot of assumptions, which are often incorrect. As a new user, please be humble and accept that you might not know everything there's to know about the software just yet. This issue is about ODM, not WebODM. Instead of being confrontational and defending your actions, please take a step back, follow the rules of our community, follow the indications of people that have contributed on this project for longer than you have and we can all continue to improving this beautiful software.

smathermather, 2023-11-10:
Superceded by https://github.com/OpenDroneMap/ODM/pull/1720

closed 2023-11-10 by smathermather