OpenDroneMap/ODM #1721: Possible bug in determination of coordinate system when using GCPs

smathermather, 2023-11-13:
Possible bug: when my data are fed in UTM Zone 36, but ODM calculates 40S as my coordinate system somehow:
```
[INFO]    Loading 30 images
[INFO]    Wrote images database: /var/www/data/01c274ba-6ef4-4ef9-8242-058902d88be0/images.json
[INFO]    Found 30 usable images
[INFO]    15 GCP points will be used for georeferencing
[INFO]    Generated coords file from GCP: WGS84 UTM 40S
```

GCP as follows:
```
+proj=utm +zone=36 +south +ellps=WGS84 +datum=WGS84 +units=m +no_defs
706784 411645 0 1810 2184 IMG_4389.jpeg
706784 411673.5 0 1704 2191 IMG_4389.jpeg
707121.5 411673.5 0 1662 1037 IMG_4389.jpeg
706784 411645 0 1145 1172 IMG_4393.jpeg
706784 411673.5 0 1139 977 IMG_4393.jpeg
707121.5 411673.5 0 3218 962 IMG_4393.jpeg
706784 411645 0 1211 1369 IMG_4394.jpeg
706784 411673.5 0 1196 1199 IMG_4394.jpeg
707121.5 411673.5 0 3168 1167 IMG_4394.jpeg
706784 411645 0 1274 918 IMG_4395.jpeg
706784 411673.5 0 1278 751 IMG_4395.jpeg
707121.5 411673.5 0 3140 757 IMG_4395.jpeg
706784 411645 0 1419 1201 IMG_4396.jpeg
706784 411673.5 0 1415 1020 IMG_4396.jpeg
707121.5 411673.5 0 2895 998 IMG_4396.jpeg
```

Images are GPS free, so the only cue for coordinate system should be the proj string in the GCP dataset, but somehow ODM is assigning a different UTM zone.

Image and GCP data here:
https://github.com/smathermather/ioniq5/

pierotofy, 2023-11-14:
Is it possible the coordinates are incorrect?

[image]

smathermather, 2023-11-15:
The coordinates in that GCP file should place it in Kenya.

This would apply even if it were a real dataset (but yeah: completely arbitrary choice here).

I'm a little confused though: the header is relevant to parse the meaning of the GCP values and thus the determination of appropriate UTM. We cannot discern the meaning of GCPs without parsing that header, and converting those values to lat/lon to pass to pass to a function to find UTMz zone, or am I misreading something?

Imagine a (real) scenario where images have no exif to parse, but GCPs are used where the coordinate system is typically local, not angular.

smathermather, 2023-11-21:
So, I assume the fix is a pull request parsing the header and with proj4, rewriting the `get_utm_zone_and_hemisphere_from` to calculate appropriate UTM zone? That would generalize to non-UTM headers that undoubtedly have the same challenge.

pierotofy, 2023-11-21:
I don't think this might be a bug, but rather the coordinates might just be incorrect?

`706784 411645 0 1810 2184 IMG_4389.jpeg`

[image]

Your Northing value in https://epsg.io/32736 places you in Antarctica:

[image]

smathermather, 2023-11-21:
I'm only off by an order of magnitude or so... . Closing.. :)

closed 2023-11-21 by smathermather

smathermather, 2023-11-27:
Just follow up, with correct coordinates (this time in a zone compatible with Korea) we get this beautiful result:
[image]