OpenDroneMap/ODM #1741: ODM Is not processing MP4 videos when uploaded with SRT file

DanielALT0134, 2024-01-23:
Hi,

I've been trying to process a MP4 video with the SRT file to increse the resolution, but with no success.  When I try to do it without the SRT file it works correctly.

See the log file
[console.txt](https://github.com/OpenDroneMap/ODM/files/14022485/console.txt)

If necessary, here is the [link](https://www.dropbox.com/scl/fo/zz5ct1mbtuj68ay42yw4r/h?rlkey=im0dfhry0rz3t1bdflgmbyk2m&dl=0) to download the dataset.

pierotofy, 2024-01-23:
Thanks for the report and dataset @DanielALT0134, this should be fixed as soon as https://github.com/OpenDroneMap/ODM/pull/1742/commits/9fd3bf3eddbc856511eb931613b6157606c99f6c is merged :pray:

pierotofy, 2024-01-23:
Seems to render in the right place and with the proper scale. [image]

DanielALT0134, 2024-01-24:
Thank you Piero !

closed 2024-01-29 by pierotofy