OpenDroneMap/ODM #1743: Quality Reports should contain the parameters values

joe-stroup, 2024-01-27:
I use WebODM on both a Windows desktop and on a ubuntu/docker server.

WebODM quality reports do not contain the parameter values for the task covered by the quality report.

When using quality reports to compare the results from different parameter settings, this is essential information.

smathermather, 2024-01-27:
Excellent idea. Is this something you'd be able to help with?

joe-stroup, 2024-01-27:
Off course I’ll do what I can to help. What do you propose?

One thing I can do is provide screenshots of how Pix4D presents this information.

BTW - I don’t recall seeing a response to my query about when bug fixes become available in the update routines.  If it requires a new release of WebODM, how often are new releases published?

smathermather, 2024-01-27:
An example from other software might help, but the request seems straightforward from a requirements perspective.

The intent of the question was I was curious if you had capacity to look at the python and address the issue. I'd be happy to find a relevant pull request to model any changes, if that would help.

If not, no problem. Not everyone who files an issue has the capacity, but it's the question we often ask as devs are busy.

Regardless, thanks for the feature request. It is a very useful suggestion.

(Side note: Please follow up with community forum conversations on the community forum. Don't cross post here.)

Thanks!

joe-stroup, 2024-01-27:
I appreciate your vote of confidence, but my coding days are decades ago - and not likely to spin-up again.

I'm going to send you a PM to explain a bit more.

In the mean time, I'm happy to test changes and new features - and make suggestions.

smathermather, 2024-01-27:
No problem. Thanks for opening the issue. Cheers.