OpenDroneMap/ODM #1748: Feature request / Proposal -- increase default GPS-Accuracy to 2m

smathermather, 2024-02-29:
For a couple years now, I've been of the mind that the default GPS-Accuracy under-constrains the model using GPS values, resulting in calibration issues which presents as excessive doming in the elevation data produced. Based on processing dozens of datasets on the community forum, the accuracy (or in this case, really a measure of consistency) for individual flights (rather than aggregate datasets) can be safely tuned to 0.2m / 2 decimeters.

### Doming in dataset from Monrovia, Liberia
[image]

For aggregate (multi-flight) datasets where there can be GPS consistency differences over time, it's safer to set this value to a larger value such as 2m.

I'm currently testing the switch to 2m with roughly a dozen community datasets shared with me over the last few years:
[image]

Thoughts? Any good counter examples? I have yet to break a dataset by reducing to 2m, but I could be missing salient examples.

smathermather, 2024-02-29:
Community conversation on this here:
https://community.opendronemap.org/t/default-gps-accuracy/19510

smathermather, 2024-03-03:
In the dozen sets I tried, not a single failure at 2m. Sounds like @Saijin-Naib uses 3 and has yet to see a failure.

Saijin-Naib, 2024-03-03:
The only time I need to relax it is when using ground-level mobile phone collects with my TeraCube, but even the default 10m isnt nearly coarse enough. 30m is normally needed

smathermather, 2024-03-09:
https://github.com/OpenDroneMap/ODM/pull/1749

closed 2024-03-09 by pierotofy

smathermather, 2024-03-09:
Edited to 3m on @Saijin-Naib's recommendation.