OpenDroneMap/ODM #1750: Feature Request: deprecate lens fisheye for lens fisheye-opencv [ODM v4.0]

Saijin-Naib, 2024-03-09:
### How did you install ODM? (Docker, installer, natively, ...)?

All

### What is the problem?

fisheye-opencv should be far more robust and, ideally, will make an easy/smart upgrade without causing regressions.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

improved reconstruction

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

side-by-side comparisons can be tested with sample Parrot Bebop2 dataset. We might need to solicit more.

APOS80, 2024-03-10:
I’ve compared fisheye and fisheye-OpenCV results from an DJI Osmo Action 4 and a 180degree fisheye lens on slr.

My conclusion is that fisheye-OpenCV produce more accurate results.

smathermather, 2024-03-12:
Possible alternative until 4.0 is to just keep the signature the same, but effectively remove the lesser of the fisheye approaches:
https://github.com/OpenDroneMap/ODM/pull/1751

Not attached to this pull request, but it could have a good impact.