OpenDroneMap/ODM #1755: MultiSpec imager processing determining single band image is RGB, ignores RGB composite if present

nlraley, 2024-04-17:
In an attempt to get around the tiling and build overviews for a multi spec image set without the blue band present:

`ERROR 6: PNG driver doesn't support 6 bands.  Must be 1 (grey),
2 (grey+alpha), 3 (rgb) or 4 (rgba) bands.`

Which I believe to be the root cause for the tiles not being rendered when processing a dataset with only the R, G, NIR, and RedEdge bands present.  

When such a dataset is processed, tiles are not rendered.  Within the tiles output folder, it's creating all the folders, but no output images for the tiles are generated, and I believe the tile creation process is spitting out the errors listed above and thus not returning any output.

In an attempt to meet the band requirements for the reported bug, I attempted to extract the Blue band from the RGB composite image acquired from the regular camera using a gdal_translate to extract only the blue band.  The gdal_info for the exported image lists:

`Driver: GTiff/GeoTIFF
Files: /workspaces/flight-processor/imagery/ad88970e-aa74-410a-bf70-e4f1a93d2a3d/images/DJI_20240229130726_0006_MS_B.TIF
Size is 5280, 3956
Metadata:
  EXIF_ApertureValue=(6.64)
  EXIF_BodySerialNumber=1581F5FKD233X00DD6FU
  EXIF_ColorSpace=1
  EXIF_ComponentsConfiguration=0x01 0x02 0x03 0x00
  EXIF_Contrast=0
  EXIF_CustomRendered=0
  EXIF_DateTime=2024:02:29 13:07:26
  EXIF_DateTimeDigitized=2024:02:29 13:07:26
  EXIF_DateTimeOriginal=2024:02:29 13:07:26
  EXIF_DeviceSettingDescription=0x00 0x00 0x00 0x00
  EXIF_DigitalZoomRatio=(1)
  EXIF_ExifVersion=0230
  EXIF_ExposureBiasValue=(0)
  EXIF_ExposureMode=0
  EXIF_ExposureProgram=4
  EXIF_ExposureTime=(0.01)
  EXIF_FileSource=0x03
  EXIF_Flash=0
  EXIF_FlashpixVersion=0100
  EXIF_FNumber=(10)
  EXIF_FocalLength=(12.29)
  EXIF_FocalLengthIn35mmFilm=24
  EXIF_GainControl=0
  EXIF_GPSAltitude=(225.327)
  EXIF_GPSAltitudeRef=0x00
  EXIF_GPSDateStamp=2024:02:29
  EXIF_GPSLatitude=(38) (56) (6.618)
  EXIF_GPSLatitudeRef=N
  EXIF_GPSLongitude=(84) (47) (32.4322)
  EXIF_GPSLongitudeRef=W
  EXIF_GPSMapDatum=WGS-84
  EXIF_GPSStatus=A
  EXIF_GPSTimeStamp=(19) (7) (46)
  EXIF_GPSVersionID=0x02 0x03 0x00 0x00
  EXIF_ImageDescription=default
  EXIF_Interoperability_Index=R98
  EXIF_Interoperability_Version=0x30 0x31 0x30 0x30
  EXIF_ISOSpeedRatings=110
  EXIF_LensSpecification=(0.24) (0.24) (2.8) (11)
  EXIF_LightSource=1
  EXIF_Make=DJI
  EXIF_MakerNote=DJI MakerNotes
  EXIF_MaxApertureValue=(2.97)
  EXIF_MeteringMode=1
  EXIF_Model=M3M
  EXIF_Orientation=1
  EXIF_PixelXDimension=5280
  EXIF_PixelYDimension=3956
  EXIF_ResolutionUnit=2
  EXIF_Saturation=0
  EXIF_SceneCaptureType=0
  EXIF_SceneType=0x01
  EXIF_SensitivityType=2
  EXIF_Sharpness=0
  EXIF_ShutterSpeedValue=(6.6439)
  EXIF_Software=11.05.02.05
  EXIF_SubjectDistance=(0)
  EXIF_WhiteBalance=0
  EXIF_XResolution=(72)
  EXIF_YCbCrPositioning=2
  EXIF_YResolution=(72)
Image Structure Metadata:
  INTERLEAVE=BAND
Corner Coordinates:
Upper Left  (    0.0,    0.0)
Lower Left  (    0.0, 3956.0)
Upper Right ( 5280.0,    0.0)
Lower Right ( 5280.0, 3956.0)
Center      ( 2640.0, 1978.0)
Band 1 Block=5280x1 Type=Byte, ColorInterp=Blue`

So this is clearly a single band, with the band color interpretation reporting as Blue.  However, when adding these blue band images to the workspace, ODM identifies them as RGB despite only having the single band.  It appears the ODM processing is only looking for the:

`band_name = self.get_xmp_tag(xtags, ['Camera:BandName', '@Camera:BandName', 'FLIR:BandName'])`

to determine the band name, and this information is not present.  It seems that in this scenario, it's treating the single band blue image as an RGB, instead of a single banded image.

If I attempt to place the RGB composite image in there with the images from the different sensors, i.e. the Red, Green, RedEdge, and NIR, I receive a message trying to process that there's duplicate bands, and it tosses out the RGB band, instead of identifying that there's no blue band present, and using the blue band data from the RGB set as the source for the blue band.

Is there any way to work around this to get it to process the tiles for the multi spec imagery?  

First, I thought adding the RGB composite to the workspace would resolve this, since it includes the blue band data, but it ignores these b/c the red and green band spec images are present.  Then, I thought I could just extract the blue band from the RGB comp and add them into the workspace would get by the restriction, but the process isn't treating them as single band.  I even added the XMP tag to get it to treat it as a Blue Band, but the processing will crash with these present.

Thanks

nlraley, 2024-04-17:
I realize this is probably an issue with GDAL, since the gdal tiling doesn't work under this scenario itself.

pierotofy, 2025-02-25:
As mentioned, this looks like a problem with GDAL (or of its use).

closed 2025-02-25 by pierotofy