OpenDroneMap/ODM #1759: Feature request: moar tag types for DOP

smathermather, 2024-05-10:
I could be misreading, but as far as I can tell, a limited number of DOP tag estimates are used. The DJI Mavic 3EM writes in the following form:

```
Rtk Std Lon                     : 1.23673
Rtk Std Lat                     : 1.19709
Rtk Std Hgt                     : 3.91458
```

Propose we build in logic to support this and other formats for tags. Happy to update get_gps_dop to support more tag types, but figured I'd open an issue first since it's friday and my brain is mush.

pierotofy, 2024-05-10:
https://github.com/OpenDroneMap/ODM/blob/master/opendm/photo.py#L387 ?

smathermather, 2024-05-10:
Looking in the wrong location... . Thanks!

closed 2024-05-10 by smathermather

smathermather, 2024-05-10:
Todo: GitHub that auto closed any issues I open on Friday afternoon.