OpenDroneMap/ODM #1763: MacOS native install: No formulae or casks found for tbb@2020

kevinpauli, 2024-06-06:
### How did you install ODM? (Docker, installer, natively, ...)?

natively on mac

### What is the problem?

This line in configure_macos.sh:

```
brew install cmake gcc@12 python@3.8 tbb@2020 eigen gdal boost cgal libomp
```

Is giving this error:
```
Warning: No available formula with the name "tbb@2020".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for tbb@2020.
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

It should configure macos correctly

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

```
git clone https://github.com/OpenDroneMap/ODM
cd ODM
bash configure_macos.sh install
```

pierotofy, 2025-01-04:
> I guess this fits into the business model of charging for an installer/compiled binary and not giving support to manually install.

This has nothing to do with the business model, and lots to do with the complexity that is building ODM from source. We recommend people use docker to run ODM on macOS, which doesn't cost anything and works out of the box.

smathermather, 2025-01-05:
> We recommend people use docker to run ODM on macOS, which doesn't cost anything and works out of the box.

As Piero indicates, @kevinpauli @JensTimmerman, please use the [README](https://github.com/OpenDroneMap/ODM?tab=readme-ov-file#quickstart) or docs.opendronemap.org for directions on how to install and use.

I will close this issue for now. If anyone has a pull request to address native mac install, we can re-open as appropriate.

smathermather, 2025-01-05:
> Docker works if you are willing to/able to install docker on your machine. (Requires admin rights e.g) to run the application

Yes, there are limitations to docker, e.g. admin rights. It is possible to run within other containerization that don't require the same escalation, e.g. apptainer. Docker is the recommended approach at this time, but feel free to test and write documentation for use in other container systems as appropriate.

> But what if you want to create your own modifications and run those?

This is beyond the scope of this issue to document, but I tend to use a workflow similar to this: https://docs.opendronemap.org/tutorials/#development-and-testing-of-odm

> directing to the documentation is a bit weird.
> the issue arrived from following the steps in the readme
> https://github.com/OpenDroneMap/ODM?tab=readme-ov-file#native-install-macos

Cool. That wasn't made clear in the initial issue, so I missed that was what was being referenced. Thanks for linking directly to it to clarify the need.

A pull request updating with a good or updated workflow would be most appreciated. Alternatively a pull request with an update to the readme to deprecate or warn in the native section would be a second best solution.

> And the software stack used on the docker images is completely out of date and no longer supported.

If you are willing to take on stack updates in a serious way, let us know.

dokterbob, 2025-04-03:
This issue is still present, after following basic steps in the documentation. 

I would suggest to either officially drop native Mac builds (e.g. remove from README) _or_ re-open and to leave this open in the form of a tracking ticket.