OpenDroneMap/ODM #1778: Asking for explanation on why using '@drone-dji:FlightYawDegree' instead of '@drone-dji:GimbalYawDegree'

khaled-hammoud, 2024-07-12:
I am testing the source code here then I  @came to this line:

https://github.com/OpenDroneMap/ODM/blob/a6bcb28fbdcf6ff707ee438ad3c8443bb2be7d39/opendm/photo.py#L452

when converting from ypr to omk, why to use gimbal's pitch and roll but flight's yaw eventhough the gimbal's yaw data is also available in exif for DJI drone for example MA2.

I checked dozens of images that I captured then compared `FlightYawDegree` with `GimbalYawDegree` and found differences not 1 or 2 degrees but sometimes more.

So I am confused why to use flight yaw instead of gimbal's yaw, the camera is on gimbal!
Could someone explain for me this thing, thanks!

pierotofy, 2024-07-12:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2024-07-12 by pierotofy

khaled-hammoud, 2024-07-12:
Sorry! I am new here, of course!
I asked this question there where you recommended.