OpenDroneMap/ODM #1781: Possible bug in using image_groups.txt with WebODM

smathermather, 2024-07-21:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker, on Ubuntu.

### What is the problem?

Including an image_groups.txt in upload to WebODM, the txt file is correctly identified and copied into the appropriate directory:

> [INFO]    split_image_groups: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/gcp/image_groups.txt

> [INFO]    Copied /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/gcp/image_groups.txt to /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/opensfm/image_groups.txt

but the `split_image_groups_is_set` flag doesn't appear to be set or otherwise is bypassed:

```
[INFO]    Running split stage
[INFO]    Normal dataset, will process all at once.
[INFO]    Finished split stage
[INFO]    Running merge stage
[INFO]    Normal dataset, nothing to merge.
[INFO]    Finished merge stage
[INFO]    Running opensfm stage
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

If an image_groups.txt file is uploaded through WebODM, we should be setting split_image_groups_is_set so we proceed with split merge.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

[[Dataset here]](https://drive.google.com/drive/folders/16j5nyuizdx55y1gzVJnQDlcF-WyYf7tu?usp=sharing)

Full(enough) log as below:

```
[INFO]    Initializing ODM 3.5.3 - Sun Jul 21 20:36:50  2024
[INFO]    ==============
[INFO]    3d_tiles: False
[INFO]    align: None
[INFO]    auto_boundary: True
[INFO]    auto_boundary_distance: 0
[INFO]    bg_removal: False
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    dsm: True
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: dspsift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gltf: True
[INFO]    gps_accuracy: 3
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 0
[INFO]    matcher_order: 0
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 24
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 10000
[INFO]    name: 17369f45-d1dd-403c-ae1d-0135ed615256
[INFO]    no_gpu: False
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_copc: False
[INFO]    pc_csv: False
[INFO]    pc_ept: True
[INFO]    pc_filter: 5
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_skip_geometric: False
[INFO]    primary_band: auto
[INFO]    project_path: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256
[INFO]    radiometric_calibration: none
[INFO]    rerun: None
[INFO]    rerun_all: False
[INFO]    rerun_from: ['dataset', 'split', 'merge', 'opensfm', 'openmvs', 'odm_filterpoints', 'odm_meshing', 'mvs_texturing', 'odm_georeferencing', 'odm_dem', 'odm_orthophoto', 'odm_report', 'odm_postprocess']
[INFO]    rolling_shutter: False
[INFO]    rolling_shutter_readout: 0
[INFO]    sfm_algorithm: incremental
[INFO]    sfm_no_partial: False
[INFO]    skip_3dmodel: False
[INFO]    skip_band_alignment: False
[INFO]    skip_orthophoto: False
[INFO]    skip_report: False
[INFO]    sky_removal: False
[INFO]    sm_cluster: None
[INFO]    sm_no_align: False
[INFO]    smrf_scalar: 1.25
[INFO]    smrf_slope: 0.15
[INFO]    smrf_threshold: 0.5
[INFO]    smrf_window: 18.0
[INFO]    split: 999999
[INFO]    split_image_groups: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/gcp/image_groups.txt
[INFO]    split_overlap: 150
[INFO]    texturing_keep_unseen_faces: False
[INFO]    texturing_single_material: False
[INFO]    texturing_skip_global_seam_leveling: False
[INFO]    tiles: False
[INFO]    use_3dmesh: False
[INFO]    use_exif: False
[INFO]    use_fixed_camera_params: False
[INFO]    use_hybrid_bundle_adjustment: False
[INFO]    video_limit: 500
[INFO]    video_resolution: 4000
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/images
[INFO]    Loading 649 images
[INFO]    Found 649 usable images
[INFO]    Finished dataset stage
[INFO]    Running split stage
[INFO]    Normal dataset, will process all at once.
[INFO]    Finished split stage
[INFO]    Running merge stage
[INFO]    Normal dataset, nothing to merge.
[INFO]    Finished merge stage
[INFO]    Running opensfm stage
[INFO]    Copied /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/gcp/image_groups.txt to /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/opensfm/image_groups.txt
[INFO]    Maximum photo dimensions: 4000px
[INFO]    Photo dimensions for feature extraction: 2000px
[INFO]    Altitude data detected, enabling it for GPS alignment
```

smathermather, 2024-07-21:
But, if I'm honest, I don't really see a flaw in the logic here:
https://github.com/smathermather/ODM/blob/split-fix/stages/splitmerge.py#L34

pierotofy, 2024-07-22:
I think one needs to set `--split` with the current logic to trigger split-merge (regardless of the image_groups.txt presence). This has always been the case.

It could be changed of course.

smathermather, 2024-07-22:
I forget this every time... (every few years in a fever...).

For any that find this later, logic requires it set to a value smaller than the number of images as just setting it isn't enough:

```
[INFO]    Initializing ODM 3.5.3 - Mon Jul 22 00:47:34  2024
[INFO]    ==============
[INFO]    3d_tiles: False
[INFO]    align: None
[INFO]    auto_boundary: True
[INFO]    auto_boundary_distance: 0
[INFO]    bg_removal: False
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    dsm: True
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: dspsift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gltf: True
[INFO]    gps_accuracy: 3
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 0
[INFO]    matcher_order: 0
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 24
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 10000
[INFO]    name: 17369f45-d1dd-403c-ae1d-0135ed615256
[INFO]    no_gpu: False
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_copc: False
[INFO]    pc_csv: False
[INFO]    pc_ept: True
[INFO]    pc_filter: 5
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_skip_geometric: False
[INFO]    primary_band: auto
[INFO]    project_path: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256
[INFO]    radiometric_calibration: none
[INFO]    rerun: None
[INFO]    rerun_all: False
[INFO]    rerun_from: ['dataset', 'split', 'merge', 'opensfm', 'openmvs', 'odm_filterpoints', 'odm_meshing', 'mvs_texturing', 'odm_georeferencing', 'odm_dem', 'odm_orthophoto', 'odm_report', 'odm_postprocess']
[INFO]    rolling_shutter: False
[INFO]    rolling_shutter_readout: 0
[... 131 lines cut ...]
[INFO]    use_fixed_camera_params: False
[INFO]    use_hybrid_bundle_adjustment: False
[INFO]    video_limit: 500
[INFO]    video_resolution: 4000
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/images
[INFO]    Loading 649 images
[INFO]    Found 649 usable images
[INFO]    Finished dataset stage
[INFO]    Running split stage
[INFO]    Large dataset detected (649 photos) and split set at 648. Preparing split merge.
[INFO]    Copied /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/gcp/image_groups.txt to /var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/opensfm/image_groups.txt
[INFO]    Maximum photo dimensions: 4000px
[INFO]    Photo dimensions for feature extraction: 2000px
[INFO]    Altitude data detected, enabling it for GPS alignment
[INFO]    ['use_exif_size: no', 'flann_algorithm: KDTREE', 'feature_process_size: 2000', 'feature_min_frames: 10000', 'processes: 24', 'matching_gps_neighbors: 0', 'matching_gps_distance: 0', 'matching_graph_rounds: 50', 'optimize_camera_parameters: yes', 'reconstruction_algorithm: incremental', 'undistorted_image_format: tif', 'bundle_outlier_filtering_type: AUTO', 'sift_peak_threshold: 0.066', 'align_orientation_prior: vertical', 'triangulation_type: ROBUST', 'retriangulation_ratio: 2', 'matcher_type: FLANN', 'feature_type: DSPSIFT', 'use_altitude_tag: yes', 'align_method: auto', 'local_bundle_radius: 0', 'submodels_relpath: ../submodels/opensfm', 'submodel_relpath_template: ../submodels/submodel_%04d/opensfm', 'submodel_images_relpath_template: ../submodels/submodel_%04d/images', 'submodel_size: 648', 'submodel_overlap: 150']
[INFO]    Wrote reference_lla.json
[INFO]    running "/code/SuperBuild/install/bin/opensfm/bin/opensfm" detect_features "/var/www/data/17369f45-d1dd-403c-ae1d-0135ed615256/opensfm"
2024-07-22 00:50:15,229 INFO: Planning to use 439106.080078125 MB of RAM for both processing queue and parallel processing.
2024-07-22 00:50:15,230 INFO: Scale-space expected size of a single image : 90.83747863769531 MB
2024-07-22 00:50:15,230 INFO: Expecting to queue at most 200 images while parallel processing of 24 images.
```

So, not a bug, but definitely a rough edge. I've got a pull request coming. It's probably wrong, but worth a try.

smathermather, 2024-07-22:
https://github.com/OpenDroneMap/ODM/pull/1782

closed 2024-07-22 by smathermather