OpenDroneMap/ODM #1785: Feature request: option to disable OpenMVS point cloud filter (filtering by visibility)

originlake, 2024-07-30:
I noticed the OpenMVS point cloud filter might have a problem, how it determines the point visibility cannot handle when seeing a plane from side. I had a dataset that the majority of good points on a plane get removed. 

Before filter
[image]

After filter
[image]

I believe the case like the red camera is causing this issue, when seeing a plane from side, every closer point occludes further points, which leads to unpredictable behavior in the visibility filtering algorithm. This might also cause points on the wall of a building removed. I can share the intermediate openmvs files if needed (scene_dense.mvs), images won't be allowed.
[image]

pierotofy, 2024-07-31:
Interesting find, as a workaround there's the option to skip filtering in ODM by setting `pc-filter: 0`, but would be awesome to improve things in OpenMVS.

smathermather, 2024-08-01:
@originlake -- thanks for getting this issue into the OpenMVS queue too.