OpenDroneMap/ODM #1787: `[ERROR]   Directory project does not exist` when following README.md using `docker run -ti --rm -v /home/youruser/datasets:/datasets opendronemap/odm --project-path /datasets project`

moefear85, 2024-08-02:
### How did you install ODM? (Docker, installer, natively, ...)?

docker

### What is the problem?

docker isn't running. output: [ERROR]   Directory project does not exist

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

expected to run

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

just type docker run -ti --rm -v /home/youruser/datasets:/datasets opendronemap/odm --project-path /datasets project from home directory, replacing `youruser` with your username.

pierotofy, 2024-08-02:
This doesn't look like a bug; if you need help running ODM, let's move the conversation over to https://community.opendronemap.org? 🙏

closed 2024-08-02 by pierotofy

moefear85, 2024-08-02:
maybe so, but it requires modifying the readme.md hence github tree to provide a working docker statement, since the presence of it implies that it is considered relevant that a simple docker statement should work out-of-the-box.