OpenDroneMap/ODM #1793: Feature suggestion: adding TIFFTAG_DATETIME to .tif outputs

stephenwinn16, 2024-08-13:
Hi, 

I'm looking into adding some metadata to the `.tif` outputs of ODM such that I have access to this information in a self-contained manner. Specifically, I would like the file metadata to include some estimate of the capture time (e.g. the mean of capture start/capture end times). I see [here](https://gdal.org/drivers/raster/gtiff.html) that geotiff files support the TIFFTAG_DATETIME tag (I use this info to e.g. get the capture time of satellite data I'm working with). Do you think it would be a case of adding an `-mo` argument to the GeoTIFF creation stage e.g. as is done below for the `-co` arguments? 

https://github.com/OpenDroneMap/ODM/blob/e6e4095844017d4f84d7d845e4d14780059703f6/stages/odm_orthophoto.py#L102

As well as something similar for the DSM/DTM creation steps if they happen. I'm happy to take a shot at adding this to the source code if I'm on the right track here.

pierotofy, 2024-08-13:
I think that could be useful and a pull request would be most welcome.

stephenwinn16, 2024-09-06:
Implemented in PR #1797.

closed 2024-09-06 by stephenwinn16