OpenDroneMap/ODM #1794: Inconsistent Orthomosaic Image Results Using OpenDroneMap with PyODM

mithilpatil1994, 2024-08-14:
I'm using PyODM to generate orthomosaic images from a large set of TIFF files (ranging from 1,000 to 5,000 images) captured by a drone. However, I notice that the generated orthomosaic image is different each time I run the process, even though I'm using the same set of images and the same code.

### Setup:
- **Repository**: I'm using the OpenDroneMap repository cloned as follows:
    ```bash
    # Clone the OpenDroneMap repository
    git clone https://github.com/OpenDroneMap/ODM --recursive
    cd ODM
    ```
- **Code**: Here’s a simplified version of my Python code using PyODM:

```python
import threading
from pyodm import Node, exceptions
import glob
import os
from pathlib import Path

def progress_callback(task_id):
    if int(task_id == 100):
        print("Uploading Images done")

def print_progress(task):
    taskInfo = task.info()
    print(taskInfo.status)

def startOrthoProcess():
    node = Node('localhost', port=3000)
    try:
        print(node.info())
    except exceptions.NodeConnectionError as e:
        print("Cannot connect: " + str(e))
        return {
            "status": 500,
            "message": "Cannot connect to Node"
        }

    option_flags = {
        'auto-boundary': True, 
        'radiometric-calibration': "camera"
    }
    
    image_files = [str(p) for p in Path("/path/to/images").rglob("*.tif")]
    print(len(image_files))
    
    task = node.create_task(image_files, option_flags, progress_callback=progress_callback)
    taskInfo = task.info()
    print(taskInfo)
    
    timer = threading.Timer(60.0, print_progress, args=[task])  # 60 second timer to show progress
    timer.start()
    
    try:
        result_dir = "results"
        if not os.path.exists(result_dir):
            os.makedirs(result_dir)
        
        task.wait_for_completion()
        task.download_assets(result_dir)
        print(f"Result saved in: {result_dir}/odm_orthophoto/odm_orthophoto.tif")
        
        return {
            "status": 200,
            "message": "Successfully generated Orthomosaic"
        }

    except exceptions.TaskFailedError as e:
        print("Error: " + str(e))
        return {
            "status": 500,
            "message": str(e)
        }

def main():
    startOrthoProcess()

if __name__ == "__main__":
    main()
```

### What I've Tried:
- I've ensured that the images and metadata are consistent.
- The `option_flags` are the same for each run.
- I've checked the server environment, and it seems stable.
- The variability persists even when processing 1,000 to 5,000 images.

### Questions:
- What could cause the orthomosaic images to differ each time the process is run?
- Are there any specific options or flags in OpenDroneMap that might introduce variability?
- How can I ensure that the results are consistent across multiple runs?

Any insights would be greatly appreciated!

closed 2024-08-14 by github-actions[bot]

pierotofy, 2024-08-14:
Also, for all questions, let's move the conversation over to https://community.opendronemap.org :pray:

smathermather, 2024-08-14:
Thanks for joining us on the forum!

Linking back to the forum in case anyone finds this through a search of issues:
https://community.opendronemap.org/t/inconsistent-orthomosaic-image-results-using-opendronemap-with-pyodm-1794/21393