OpenDroneMap/ODM #1798: False out of memory error.

John-Nagle, 2024-08-23:
### How did you install ODM? (Docker, installer, natively, ...)?

As a snap on Ubuntu 22.04 LTS

> opendronemap --version
ODM 3.4.0

Tested the install by running the "brighton beach" test images, which worked.

### What is the problem?

```
Estimated depth-maps 108 (75.52%, 4m23s, ETA 1m)...terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.0) /build/opendronemap/parts/odm/build/SuperBuild/src/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 66362010468 bytes in function 'OutOfMemoryError'"

```
Note the number of bytes. Something tried to allocate 66 gigabytes. Memory consumption was only 4GB (on a 32 GB machine) according to the system memory usage graph (attached) up to that point.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Not crashing.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Get the data set here: https://animats.com/sharpview/testdata/ 
Get the .zip file from the link "Port Babbage 2".

Log:  [log.json](https://github.com/user-attachments/files/16724263/log.json)

Console output: 
[falseoomcrash.txt](https://github.com/user-attachments/files/16724265/falseoomcrash.txt)

Memory usage graph while ODM running:

[image]

### Notes
Open Drone Map was called by:

`opendronemap --project-path .
`
This is a test of impostor generation for a virtual world. Input is a set of roughly 100 images from a flyover. It's an unusual use of
ODM, but shouldn't crash it. The images are reasonable looking. 

It looks like the matching phase went well, but then, 75% through depth matching, something requested an unreasonable amount of memory all at once.

John-Nagle, 2024-08-23:
Attempt to rerun from the results of the previous run produced the same error message immediately. So it's reproduceable.

pierotofy, 2024-08-23:
Does it work with docker? We no longer support snap. See https://github.com/OpenDroneMap/ODM/issues/1709

John-Nagle, 2024-08-23:
I provided the image set. Did it run with a later version of Open Drone Map?

pierotofy, 2025-02-24:
I finally had a chance to run this dataset; the reason it uses so much memory, is because these are screenshots from a virtual world, and the skymap sphere trips the software into triangulating points near infinity.

[image]

You might need to create image masks (https://docs.opendronemap.org/masks/) to block the sky/background in order to process this. Otherwise, it will allocate a huge amount of memory.

closed 2025-02-24 by pierotofy

John-Nagle, 2025-02-24:
Having to manually do an edit for each image is not very helpful. 

I can have any surround I want, though. What can I use that will make ODM just ignore the thing.

pierotofy, 2025-02-24:
Set it to a neutral color like white or black.

John-Nagle, 2025-02-24:
OK.