OpenDroneMap/ODM #1804: SMRF replacement?

smathermather, 2024-09-26:
Call me grump, but I don't like SMRF. I think it's difficult to tune, still doesn't give great results, and could be upgraded / updated with better algorithms easily.

[image]

Any objection to testing the alternatives? And if we are doing so, what are our criteria?

https://pdal.io/en/latest/stages/filters.html#classification

I'm a huge fan of filters.pmf in my own work, mostly just set and forget. What have others experienced?

luc2323, 2025-01-02:
https://www.tandfonline.com/doi/full/10.1080/17538947.2020.1791267#d1e363

Have had good results with the SMRF parameters suggested in this paper. Have not tested the others.

smathermather, 2025-01-05:
Very interesting. Anything you can share here?

lucadelu, 2025-02-18:
I'm in flavor to add filters.pmf as option, but not to remove smrf.

my2cents