OpenDroneMap/ODM #1805: GPS position not available, The orthophoto will not be georeferenced.

sunmiaobo, 2024-10-11:
### How did you install ODM? (Docker, installer, natively, ...)?

Installer (Windows 10)  odm3.5.4

### What is the problem?

input DJI air 2 video and srt .

GPS position not available.  
Could not generate coordinates file. The orthophoto will not be georeferenced.

.srt formats :
1
00:00:00,000 --> 00:00:00,016
<font size="36">SrtCnt : 1, DiffTime : 16ms
2024-04-19 14:46:07,308,705
[iso : 100] [shutter : 1/180.0] [fnum : 280] [ev : 0] [ct : 5139] [color_md : default] [focal_len : 319] [latitude: 39.778940] [longitude: 117.004615] [altitude: 26.600000] 
</font>

pierotofy, 2024-10-18:
Can you upload a copy of your dataset? (video+srt).

sunmiaobo, 2024-10-21:
> Can you upload a copy of your dataset? (video+srt).

[images.zip](https://github.com/user-attachments/files/17456274/images.zip)
part of dataset

pierotofy, 2024-10-21:
Thanks! Seems to be a bug due to an additional `</font>` tag in the SRT file which was not expected. Fix coming soon..

PR #1808 (2024-10-21): Fix RST parsing for DJI air 2
Fixes an issue with SRT parsing with the DJI Air 2

Closes #1805
closed 2024-10-21 by pierotofy