OpenDroneMap/ODM #1809: Why to use FlightYawDegree instead of GimbalYawDegree

khaled-hammoud, 2024-10-28:
While I am looking the file `photo.py`, exactly here:
https://github.com/OpenDroneMap/ODM/blob/12e5b63b6f5423383892d325467b64ce295e3a9c/opendm/photo.py#L463

I see using of gimbal's roll and pitch BUT flight's yaw instead of gimbal's yaw, why?

On the other side when looking into file `exif.py` from opensfm, exactly here:
https://github.com/mapillary/OpenSfM/blob/537275b730d7a6417fce68c3645f69c834bddca3/opensfm/exif.py#L494

then I see what I expect or claim to be, is there any explanation for that?

[image]

My regards.

pierotofy, 2024-10-28:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2024-10-28 by pierotofy

khaled-hammoud, 2024-10-28:
Is that not a potential misstype to be asked here?