OpenDroneMap/ODM #1810: Add debugging symbols to build

smathermather, 2024-11-01:
I have someone helping with a testing environment for running test datasets through the pipeline when there are changes to the code base, a bit CI/CD style. @pierotofy: Is it appropriate or problematic to add debugging symbols to the default docker builds to facilitate this?

pierotofy, 2024-11-01:
I don't think that would be appropriate; rebuilding the image with debugging symbols for the specific module that is giving issues (if needed) would be better.

Or use `./start-dev-env.sh` and rebuild just the module you need with debugging symbols turned on (that's what I usually do).

smathermather, 2024-11-02:
Testing on an individual basis, that makes perfect sense. The idea would be to run in a CI/CD pipeline continuously, so having debugging symbols available in the image would be the ideal. We could possibly have a separate image build for these purposes. @polvi might be able to explain better as this is well outside my wheelhouse. This stems from a request to build out testing infrastructure that regularly runs test datasets through the pipeline.

pierotofy, 2024-11-03:
A separate image would work too.

smathermather, 2024-11-05:
> A separate image would work too.

Cool. We can do that.

pierotofy, 2025-02-23:
Status on this one?

smathermather, 2025-02-23:
Should know more next week or two. Having a gathering which will establish whether it's time to spend energy on this.