OpenDroneMap/ODM #1818: Files missing read permission

knl88, 2024-12-20:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker

### What is the problem?

The files:

-rw------- 1 root root opensfm/undistorted/openmvs/scene_dense.ply
-rw------- 1 root root opensfm/undistorted/openmvs/scene_dense_dense_filtered.ply
-rw------- 1 root root odm_meshing/odm_mesh.ply
-rw------- 1 root root odm_meshing/odm_25dmesh.ply

Are created with permission 600 so they aren't readable by others without changing permissions using the root user. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The above files should be created with permission 644 so they are readable by everyone, this is the behavior in ODM v3.1.0.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

```bash
git clone https://github.com/pierotofy/drone_dataset_brighton_beach.git
docker run -ti --rm -v $(pwd)/drone_dataset_brighton_beach:/datasets opendronemap/odm:3.5.3 --project-path /datasets .
ls -ltr drone_dataset_brighton_beach/odm_meshing/
```

pierotofy, 2025-02-23:
We haven't changed the behavior between versions, but it's likely a configuration in your filesystem settings (check your `/etc/fstab` umask value).

For example, on my machine, all new files created by the ODM container are:

```
-rw-r--r-- 1 root root    0 Feb 23 22:24 test.txt
```

```
root@9017b40193d0:/code# umask
0022
```

While ODM could check/add permissions while creating new files, I don't think ODM should be in charge of doing so. 🙏

knl88, 2025-02-24:
Agree that it wouldn't make much sense for ODM to check and change permissions ;)  But still think this is a problem that applies to some versions of ODM.  Also note that it is only the files

-rw------- 1 root root opensfm/undistorted/openmvs/scene_dense.ply
-rw------- 1 root root opensfm/undistorted/openmvs/scene_dense_dense_filtered.ply
-rw------- 1 root root odm_meshing/odm_mesh.ply
-rw------- 1 root root odm_meshing/odm_25dmesh.ply

all other files have `-rw-r--r--` as your `test.txt`. From a small test, running the command given above,  `opendronemap/odm:3.3.0` leads to missing permissions, while `opendronemap/odm:3.2.0` works correctly.

My initial guess would be that something have changed in https://github.com/cdcseacave/openMVS between builds, if that can make sense?

knl88, 2025-02-24:
Thanks, I did a test and rebuilt ODM v3.3.0 with 

```c++
h = ::open(aFileName, m, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
```

in openMVS [File.h#L471](https://github.com/knl88/openMVS/blob/ec47556c28f9bec98f1ffc1bb9818b182d575ec0/libs/Common/File.h#L471), and that looks to solve the problem. This isn't a big problem, but it's subtle enough to create some headache.

pierotofy, 2025-02-24:
Thanks for testing; I've brought this change into our fork and will be solved with https://github.com/OpenDroneMap/ODM/pull/1833

closed 2025-02-25 by pierotofy