OpenDroneMap/ODM #1825: Create border polygon

lucadelu, 2025-02-18:
During a course students told me that in CAD system to show correctly orthophoto it is needed a file with a polygon of the orthophoto's border. They also told me that proprietary software is creating this polygon and provide it with the image.

I searched a little bit but it looks like there is nothing like that right now, I can implement it, I was thinking to create a new function (and a new option in the cli) and add it after `args.orthophoto_cutline`, 

What do you think?

pierotofy, 2025-02-18:
Is `odm_georeferencing/odm_georeferenced_model.bounds.gpkg` perhaps what you need?

lucadelu, 2025-02-18:
> Is `odm_georeferencing/odm_georeferenced_model.bounds.gpkg` perhaps what you need?

No, that is the clipping area, I would like to have the square extend of the full tiff.

[image]

`odm_georeferencing/odm_georeferenced_model.bounds.gpkg` is the yellow, instead I would like to have green.

pierotofy, 2025-02-18:
If one wants the extent, then it should be extracted from the GeoTIFF (odm_orthophoto.tif). It could be added as a post-processing step in the odm_orthophoto stage. I don't think we should add more flags to generate it (if it's fast), just generate the file and place it in `odm_orthophoto/<file>`

Just curious, which proprietary software are we talking about and what's the file format of the extent file? Strange that CAD programs don't handle GeoTIFF automatically.

lucadelu, 2025-02-18:
[image]

this is the result of the code into #1828

PR #1828 (2025-03-02): added capability to generate vector polygon in DXF format to be used in AutoCAD LT
Fix #1825

Do you need any test for this?
closed 2025-03-02 by pierotofy