OpenDroneMap/ODM #1831: Add support for DNG files

pierotofy, 2025-02-23:
DNG files are a type of raw file supported by some DJI drones.

Using the `rawpy` and `exiftool` it should be trivial to convert them to JPEG prior to processing.

Saijin-Naib, 2025-02-23:
What if folks are using them for the wider dynamic range? Is it possible to pass DNG through as metadata-tagged TIFF (which internally, DNG is mostly just TIFF anyway)?

pierotofy, 2025-02-23:
Probably, I'm just unsure what people actually use DNG for. Maybe this would be a good discussion to have on the forum before implementing it.

Saijin-Naib, 2025-02-23:
> Probably, I'm just unsure what people's goal are for using DNGs. Maybe this would be a good discussion to have on the forum before implementing it.

From the Community posts and emails I've been involved with over the years about DNG, I summarize it like this:  

The folks that _know_ they are using DNGs want them for the extra radiometric resolution and lesser processing/artifacting.

The folks that _don't_ know they are using DNGs just are taking whatever files and throwing them at WebODM and just expect it to work it out.

pierotofy, 2025-02-23:
Makes sense, thanks for the insight. We should then process these as 16bit TIFFs, otherwise there's no point.

PR #1833 (2025-02-25): DNG/RAW/NEF file support
Closes #1831
closed 2025-02-25 by pierotofy