OpenDroneMap/ODM #1863: Small bug in WebODM: On completion, returns an error that clears when the page is reloaded

SarahDal, 2025-05-21:
I am running WebODM in Firefox on a linux machine with 32gb RAM and an i5-6400.

I am stitching a small sections of up to 350 images using the default settings. This runs to completion -

`Postprocessing: done (•̀ᴗ•́)و!`

and I can see the ortho thumbnail on the right hand side ..but underneath I have the text "An orthophoto could not be generated. To generate one, make sure GPS information is embedded in the EXIF tags of your images, or use a Ground Control Points (GCP) file."

Reloading WebODM puts everything back to normal and I can download the assets as expected.

Saijin-Naib, 2025-05-21:
Do those images have georef data? 

Can you post the full log and the dataset?

SarahDal, 2025-05-21:
[full log here](https://gist.github.com/SarahDal/496f83849175b3131906c5e8edb08797)

The dataset is fine, as I said, all outputs were processed correctly; it just gave an error which disappeared when I reloaded the WebODM dashboard.

smathermather, 2025-07-08:
I have actually been seeing this too. Seems fairly independent of data, WebODM 2.8.4, PopOS (Linux), Firefox 140.0.1.

I was just trying to get a screenshot of it and it doesn't happen. It might only happen on new datasets, not reruns.