OpenDroneMap/ODM #1864: COG output driver not recognized due to downgraded GDAL version

bockardo, 2025-05-23:
When running gdal_translate to convert a DSM to Cloud Optimized GeoTIFF (COG), I get the following error:

```
Output driver `COG' not recognised.
The following format drivers are configured and support output:
```
Full command:

`gdal_translate -of COG -co NUM_THREADS=48 -co BLOCKSIZE=256 -co COMPRESS=DEFLATE -co PREDICTOR=2 -co BIGTIFF=IF_SAFER -co RESAMPLING=NEAREST --config GDAL_CACHEMAX 48.65% --config GDAL_NUM_THREADS 48 /var/www/data/7725ed96-af08-4645-9bc4-656cedd49456/odm_dem/dsm.tif /var/www/data/7725ed96-af08-4645-9bc4-656cedd49456/odm_dem/dsm_cogeo.tif`

This used to work fine in a previous container build. After investigating, I found that the image now uses GDAL version 3.0.4 due to a dependency on ubuntugis/ubuntugis-unstable, whereas it previously used GDAL 3.4.3, which includes support for the COG driver.

pierotofy, 2025-07-08:
I think this happens with the GPU docker images only, which seem to have an outdated version of GDAL.

smathermather, 2025-07-21:
This is closely related to / perhaps duplicate of  https://github.com/OpenDroneMap/ODM/issues/1776

ehallein, 2025-07-22:
> @ehallein -- let me know if you need any support on this.

No problem. I'll have a look.

dhuyp, 2025-07-23:
@ehallein my last GPU run without this error was in February (personally). Good luck with the upgrade, I try it myself and it is worst with Ubuntu 24.04 (pip working differently), on 22.04, I were blocked on raterio...

ehallein, 2025-07-23:
> [@ehallein](https://github.com/ehallein) my last GPU run without this error was in February (personally). Good luck with the upgrade, I try it myself and it is worst with Ubuntu 24.04 (pip working differently), on 22.04, I were blocked on raterio...

odd, the GPU dockerfile hasn't changed since November 2024 https://github.com/OpenDroneMap/ODM/commit/36897fc0e98c27728bb43e172269a82766e0c9cd

dhuyp, 2025-07-23:
You will have to check with older version of the components like nodeodm and webodm :)

pierotofy, 2025-07-23:
I've had success using `sudo add-apt-repository ppa:ubuntugis/ubuntugis-unstable` to install a newer version of GDAL from there in the past.

pierotofy, 2025-07-23:
pypopsift does need some updates to work with CUDA 12, there's a PR open but haven't had the time to look into it: https://github.com/OpenDroneMap/pypopsift/pull/10

smathermather, 2025-07-23:
Thinking out loud a bit, but perhaps a recipe that starts with something similar to the no-GPU build but layers in NVidia without relying on their prebuilt images would be a good intermediate to a full upgrade. I am floating this without a sense of LOE or viability.

pierotofy, 2025-07-23:
>  one for fixing the build and one for upgrading the image. 

No preference, but I would expect the image upgrade to be non-trivial.

ehallein, 2025-07-23:
> I've had success using `sudo add-apt-repository ppa:ubuntugis/ubuntugis-unstable` to install a newer version of GDAL from there in the past. 
> 

I tried that already. I then noticed configure.sh was already doing it:

if [[ "$UBUNTU_VERSION" == *"20.04"* ]]; then
        echo "Enabling PPA for Ubuntu GIS"
        sudo $APT_GET install -y -qq --no-install-recommends software-properties-common
        sudo add-apt-repository -y ppa:ubuntugis/ubuntugis-unstable
        sudo $APT_GET update
    fi

This makes me wonder if the package on repository has changed?

ehallein, 2025-07-23:
> pypopsift does need some updates to work with CUDA 12, there's a PR open but haven't had the time to look into it: https://github.com/OpenDroneMap/pypopsift/pull/10

I gave it a go, but stated getting compile errors upstream in popsift. Looks like due to changes in cuda headers.

smathermather, 2025-07-23:
Just tested on a vanilla Ubuntu 20.04 and that PPA doesn't publish a newer GDAL for 20.04:

```
add-apt-repository -y ppa:ubuntugis/ubuntugis-unstable
....
apt-cache policy gdal-bin 
gdal-bin:
  Installed: (none)
  Candidate: 3.0.4+dfsg-1build3
  Version table:
     3.0.4+dfsg-1build3 500
        500 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages
```

pierotofy, 2025-07-23:
Or try this other one:  

```
*** 3.4.3+dfsg-1~focal0 500
        500 http://ppa.launchpad.net/ubuntugis/ppa/ubuntu focal/main amd64 Packages
```

smathermather, 2025-07-24:
> I've got 5070Ti now though.

This would be why it doesn't run the GPU. Until we patch / figure out, 5 won't work. 

Aha, saw this while writing:
https://github.com/OpenDroneMap/ODM/issues/1894

But if GDAL is working, I'll merge. At least one problem is solved and series folks will be covered.