OpenDroneMap/ODM #1871: Handle Quad Bayer Sensors

pierotofy, 2025-06-13:
Newer drones tend to use quad bayer sensors (discussion: https://community.opendronemap.org/t/real-difference-quality-between-a-drone-with-a-12mp-or-48mp-camera/7321/4) which create some memory usage issues for ODM in many scenarios.

ODM should have an option to automatically resize images from such sensors when necessary, probably via a camera database lookup, for the purpose of avoiding memory issues.

Saijin-Naib, 2025-06-13:
It also gives folks an unrealistic expectation about the resolution of their data, so I think sane downsampling with opt-out is a really good idea.

smathermather, 2025-06-19:
How do we determine effective resolution? I've used jpeg compressed size as a proxy, but are there formal measures we should be using?

smathermather, 2025-06-19:
Relatedly, it may impact feature extraction and matching in strange ways, so there could be value to down-sampling for SfM portion of the pipeline even when using all the pixels in texturing:
https://www.pix-pro.com/blog/quad-bayer