OpenDroneMap/ODM #1877: #1797 breaks COG

kikislater, 2025-07-01:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker

### What is the problem?

With PR: https://github.com/OpenDroneMap/ODM/pull/1797

COG are broken and not usable with external tools such as titiler, It works with ODM==3.5.3 and not with ODM>=3.5.4

```
[39m[INFO]    Post Processing[0m
[39m[INFO]    Adding TIFFTAGs to /datasets/code/PROCESSED_DATA/PHOTOGRAMMETRY/odm_orthophoto/odm_orthophoto.tif[0m
2025-06-17 14:13:56,682 WARNING: CPLE_AppDefined in /datasets/code/PROCESSED_DATA/PHOTOGRAMMETRY/odm_orthophoto/odm_orthophoto.tif: The IFD has been rewritten at the end of the file, which breaks COG layout.
```

Reported by @gouderg

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

COG works with other software no error generated by gdal in console

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Process dataset with ODM >= 3.5.4

pierotofy, 2025-07-01:
I did notice

kikislater, 2025-07-04:
Mentioned in the link below, these operations will be blocked in gdal >= 3.8 instead of creating new tif.

https://gdal.org/en/stable/drivers/raster/cog.html#update

So, imo, tags should be included during initial tif creation or not taken into account ...

pierotofy, 2025-07-08:
This should be fixed, but please double-check once the docker images have been updated? 🙏

closed 2025-07-08 by pierotofy

smathermather, 2025-07-10:
Not sure if it's corruptions on my end (quite possible) but I'm definitely seeing some errors on update:
```
webapp  | INFO Optimizing /webodm/app/media/project/330/task/3472601c-83fe-4aef-895a-fb9fb59272f1/assets/odm_orthophoto/odm_orthophoto.tif as Cloud Optimized GeoTIFF
webapp  | Input file size is 16111, 22489
webapp  | ERROR 1: ZIPDecode:Decoding error at scanline 4608
webapp  | ERROR 1: TIFFReadEncodedTile() failed.
webapp  | ERROR 1: IReadBlock failed at X offset 18, Y offset 36: TIFFReadEncodedTile() failed.
webapp  | CPLAcquireMutex: Error = 22 (Invalid argument)
webapp  | CPLReleaseMutex: Error = 22 (Invalid argument)
webapp  | 0...10...20Optimizing /webodm/app/media/project/330/task/28105fdb-f6cf-4835-a010-f631d63a62b3/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | INFO Optimizing /webodm/app/media/project/330/task/28105fdb-f6cf-4835-a010-f631d63a62b3/assets/odm_orthophoto/odm_orthophoto.tif as Cloud Optimized GeoTIFF
webapp  | Input file size is 13289, 17073
webapp  | ERROR 1: ZIPDecode:Decoding error at scanline 2304
webapp  | ERROR 1: TIFFReadEncodedTile() failed.
webapp  | ERROR 1: /webodm/app/media/project/330/task/28105fdb-f6cf-4835-a010-f631d63a62b3/assets/odm_orthophoto/odm_orthophoto.tif, band 1: IReadBlock failed at X offset 9, Y offset 28: TIFFReadEncodedTile() failed.
webapp  | 0...10...20...30...40...50..Optimizing /webodm/app/media/project/330/task/bcef0d7a-d0d7-49f8-b227-3ee194b921f4/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | Optimizing /webodm/app/media/project/330/task/270017b1-801c-451d-95de-160ecac8b60d/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | Optimizing /webodm/app/media/project/330/task/691ebd1b-86f6-4d64-912d-65ca140e4ec9/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | Optimizing /webodm/app/media/project/330/task/c39f045c-c286-42be-85df-ec6e5b5bd028/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | Optimizing /webodm/app/media/project/330/task/9c1a2a25-3d59-4efa-bf96-23d489125eff/assets/odm_orthophoto/odm_orthophoto.tif
webapp  | INFO Optimizing /webodm/app/media/project/330/task/9c1a2a25-3d59-4efa-bf96-23d489125eff/assets/odm_orthophoto/odm_orthophoto.tif as Cloud Optimized GeoTIFF
webapp  | Input file size is 16956, 21301
webapp  | ERROR 1: ZIPDecode:Decoding error at scanline 1280
webapp  | ERROR 1: TIFFReadEncodedTile() failed.
webapp  | ERROR 1: IReadBlock failed at X offset 5, Y offset 11: TIFFReadEncodedTile() failed.
webapp  | CPLAcquireMutex: Error = 22 (Invalid argument)
webapp  | CPLReleaseMutex: Error = 22 (Invalid argument)
```

pierotofy, 2025-07-11:
Can you try on another machine?

smathermather, 2025-07-11:
I'm honestly not worried this is a broader problem. I just wanted to highlight it here in case it's seen by others. With no other reports of it, it's probably just some old crusty data that's been transferred 4 or 5 times between different virtual environments.

And if someone else sees something similar, we'll have some corroboration that we can start with.