OpenDroneMap/ODM #1897: renderdem core dumped

disarticulate, 2025-07-28:
```
{
            "command": "renderdem \"/var/www/data/3f48100b-55f7-4af6-9481-763c0facdb0b/odm_georeferencing/odm_georeferenced_model.laz\" --outdir \"/var/www/data/3f48100b-55f7-4af6-9481-763c0facdb0b/odm_dem\" --output-type max --radiuses 0.04 --resolution 0.08 --max-tiles 0 --decimation 10 --classification -1 --tile-size 4096 --force ",
            "exitCode": 134,
            "output": [
                "Decimation set to 10",
                "Reading points from /var/www/data/3f48100b-55f7-4af6-9481-763c0facdb0b/odm_georeferencing/odm_georeferenced_model.laz",
                "Number of points: 19775334",
                "Classification dimension: Classification",
                "free(): invalid size",
                "Aborted (core dumped)"
            ]
        }
```
laz & log: https://drive.google.com/drive/folders/19E4B19LH0uOUqeuRZgkzn96Jv1M01Hdp?usp=drive_link

I can upload whatever else needed.

Is there a reason to maintain https://github.com/OpenDroneMap/RenderDEM when pdal can render DEMs? Internally, I run the following python function to get DEMs:
```
def export_to_dem(in_file, out_file, resolution = 5, radius = 2, window_size = 10, srid = TABLE_SRID_TAG):
  pipeline = ensure_pipeline(in_file)
  pipeline |= pdal.Writer.gdal(
    filename = out_file,
    override_srs = srid,
    window_size = window_size,
    resolution = resolution,
    output_type = 'idw',
    radius = radius
  )
  count = pipeline.execute()
```

pierotofy, 2025-07-28:
We wrote renderdem because it's much faster than PDAL.

smathermather, 2025-07-29:
If you've got a docker environment setup that would be useful for replicating, I can try too.

disarticulate, 2025-07-30:
```
  odm-node-1:
    <<: *logging_config
    network_mode: "host"
    restart: always
    image: opendronemap/nodeodm:gpu
    container_name: kclf_odm_node_1
    entrypoint:
      - /usr/bin/node
      - /var/www/index.js
      - -p
      - ${DOCKER_EXPOSE_ODM_NODE_PORT}
      - "-q 28"
      - "--max_concurrency 30"
    deploy:
      resources:
        limits:
          cpus: "28"
        reservations:
          devices:
            - driver: nvidia
              device_ids: ["0"]
              capabilities: [gpu]
```
This is what I had. I'm going to release the data with AGPLv3 license on the forums, hopefully as early as Friday. The system has 16 cores/32 threads. This is just a docker compose setup. The project is submitted with max concurrency 30, also, so not sure. when I google the exit code, it can be sent by docker to the nodes.

smathermather, 2025-07-31:
So, dem2mesh handles OOM conditions gracefully and reduces concurrency when necessary. I don't know if renderdem does offhand, but it might be beneficial to add some logic similar to dem2mesh

disarticulate, 2025-07-31:
other options would be to decode some of the exit codes for mor clarity: https://komodor.com/learn/exit-codes-in-containers-and-kubernetes-the-complete-guide/

disarticulate, 2025-08-02:
The process runs on 256GB, 16 cores, 32 threads. However, I'm doing dev work, so it's possible we are hitting limits while other things are processing.

This latest one looks like I've got errant data somewhere so it makes sense to crash:

[image]

Manually running the commands and watching cores+cpu, and changing decimation, I get:
```
odm@bossman:/var/www/data$ /code/SuperBuild/install/bin/renderdem "/var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz" --outdir "/var/www/data/000389_2025-07-29-1248-to-0132/odm_dem" --output-type max --radiuses 0.04 --resolution 0.08 --max-tiles 0 --decimation 10 --classification -1 --tile-size 4096 --force
Decimation set to 10
Reading points from /var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz
Number of points: 23360868
Classification dimension: Classification
double free or corruption (out)
Aborted (core dumped)

odm@bossman:/var/www/data$ /code/SuperBuild/install/bin/renderdem "/var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz" --outdir "/var/www/data/000389_2025-07-29-1248-to-0132/odm_dem" --output-type max --radiuses 0.04 --resolution 0.08 --max-tiles 0 --decimation 10 --classification -1 --tile-size 4096 --force
Decimation set to 10
Reading points from /var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz
Number of points: 23360868
Classification dimension: Classification
double free or corruption (out)
Aborted (core dumped)

odm@bossman:/var/www/data$ /code/SuperBuild/install/bin/renderdem "/var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz" --outdir "/var/www/data/000389_2025-07-29-1248-to-0132/odm_dem" --output-type max --radiuses 0.04 --resolution 0.08 --max-tiles 0 --decimation 50 --classification -1 --tile-size 4096 --force
Decimation set to 50
Reading points from /var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz
Number of points: 23360868
Classification dimension: Classification
munmap_chunk(): invalid pointer
Aborted (core dumped)

odm@bossman:/var/www/data$ /code/SuperBuild/install/bin/renderdem "/var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz" --outdir "/var/www/data/000389_2025-07-29-1248-to-0132/odm_dem" --output-type max --radiuses 0.04 --resolution 0.08 --max-tiles 0 --decimation 50 --tile-size 4096 --force
Decimation set to 50
Reading points from /var/www/data/000389_2025-07-29-1248-to-0132/odm_georeferencing/odm_georeferenced_model.laz
Number of points: 23360868
Classification dimension: Classification
munmap_chunk(): invalid pointer
Aborted (core dumped)
```
Obviously the laz data is messed up on the recent one, somewhere. But the link above wasn't messed up. I'll upload the messed up laz+log to the link above. I got a writeup for the forum, but I'm still developing some other stuff so it might take a bit to release the full datasets.