OpenDroneMap/ODM #1905: Bug in the NodeODM export pipeline

rufat24, 2025-08-06:
GLTF/GLB exporter fails when --texturing-single-material is enabled

Internally, the gltf export step relies on a proper textured mesh, and when single-material is active, the way materials are combined break compatibility with the GLTF export step.

As a result, the GLTF generation step silently fails or is skipped — you get no .glb file.

smathermather, 2025-08-07:
@rufat24 -- would you be able to do some testing on the pull request linked with proposed fix?

smathermather, 2025-09-02:
Can you check to see if the textures are actually used in the output, or just generated?

rufat24, 2025-09-02:
i think i found the way to solve this issue.
All u need to do is to change the order of the args check
First it should be checked for single material and then if gltf enabled it should pack it to gltf. Currently in original version(not in this pr) it is implemented vise versa.

smathermather, 2025-09-03:
Can you give me example flags in use so I can do some direct testing and comparisons?

rufat24, 2025-09-17:
@smathermather For sure that would be great